esp-rs/esp-idf-sys #336: Problems with rust-analyzer on vscode

metehus, 2024-08-29:
I'm trying to make a new project using esp-idf with rust, and I can setup and build correctly on the terminal. The only problem is that the rust analyzer extension does not work correctly, which makes it hard to develop.

For example, some stuff works, but a lot doesn't work and keep as {unknown}
[image]

The repository of the project is https://github.com/metehus/esp-rs-idf-poc

The output of the extension is the following:
```sh
2024-08-29T23:24:45.154086Z ERROR FetchBuildDataError:
The following warnings were emitted during compilation:

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/home/matehus/projects/esp-rs-idf-poc/target/debug/build/esp-idf-sys-5197497ac81f2aca/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=could not identify the root crate and `ESP_IDF_SYS_ROOT_CRATE` not specified

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'x86_64-unknown-linux-gnu'

  Stack backtrace:
     0: anyhow::error::<impl anyhow::Error>::msg
               at /home/matehus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/anyhow-1.0.86/src/backtrace.rs:27:14
     1: build_script_build::native::cargo_driver::chip::Chip::detect
               at ./build/native/cargo_driver/chip.rs:49:18
     2: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:47:27
     3: build_script_build::native::build
               at ./build/native.rs:17:9
     4: build_script_build::main
               at ./build/build.rs:51:24
     5: core::ops::function::FnOnce::call_once
               at /home/rust/rust/library/core/src/ops/function.rs:250:5
     6: std::sys_common::backtrace::__rust_begin_short_backtrace
               at /home/rust/rust/library/std/src/sys_common/backtrace.rs:155:18
     7: std::rt::lang_start::{{closure}}
               at /home/rust/rust/library/std/src/rt.rs:159:18
     8: std::rt::lang_start_internal
     9: std::rt::lang_start
               at /home/rust/rust/library/std/src/rt.rs:158:17
    10: main
    11: <unknown>
    12: __libc_start_main
    13: _start

```

Vollbrecht, 2024-08-30:
>  Error: Unsupported target 'x86_64-unknown-linux-gnu'

You need to make sure that rust-analyzer builds with the same set of configs as your normal `cargo b` is run. E.g the stuff that is setup in .cargo/config.toml.

In this case its most likely the problem that it will build with --all-targets and this cannot work. In vscode you can set stuff like
`rust-analyzer.checkOnSave.allTargets": false,` to make sure you run RA only against your current target.

ivmarkov, 2024-09-15:
Closing due to lack of activity.

closed 2024-09-15 by ivmarkov