esp-rs/esp-idf-sys #337: MCPWM_GEN_COMPARE_EVENT_ACTION_END() doesnt exist?

Violabitch5, 2024-09-08:
Hi there,

Ive looked through the open and closed issues, but it doesnt look like this has been discussed so far.

After creating a MCPWM generator, it can be fed with a list of actions via 

esp_idf_sys::mcpwm_generator_set_actions_on_compare_event

This list needs to be terminated with this macro:

MCPWM_GEN_COMPARE_EVENT_ACTION_END()

which I cannot find anywhere in the crate. Also, the existing actions are just constants from 0 to 3. So just feeding a 0 at the end wont work either.

Workaround is obviously to use the function which only feeds one action at a time:

esp_idf_sys::mcpwm_generator_set_**action**_on_compare_event

I guess its not that of a big deal, but it just leads to more boilerplate code when a longer list is used.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/mcpwm.html#mcpwm-generator-actions-on-events
[image]

ivmarkov, 2024-09-08:
`bindgen` is limited in that it cannot expose to Rust any C macro but the most simple ones which define constants.
Other than re-implementing the equivalent of `MCPWM_GEN_COMPARE_EVENT_ACTION_END` (by looking at the ESP IDF source code) I don't have other "easy" ideas.

This is also what we do when we have to reach to ESP IDF function macros...

A more complex idea would be for `esp-idf-sys` to somehow have a whole new machinery - in the form of abilities to compile `.c` files which are separate from ESP IDF itself, and then wrap a bunch of hard-coded pre-defined macros from ESP IDF as C functions (which you can then call from Rust). Certainly possible, but a lot of work.

Violabitch5, 2024-09-08:
Ok, good to know!

Thanks a lot for the quick response guys!

closed 2024-09-15 by Vollbrecht