esp-rs/esp-idf-sys #341: Implement `esp_idf_min_version_MAJOR_MINOR_PATCH` cfgs for all released ESP IDFs so far

ivmarkov, 2024-09-18:
E.g. `#[cfg(esp_idf_min_version_5_1_2)]` would be true iff the current ESP IDF version is >= 5.1.2

Reason: [this](https://github.com/esp-rs/esp-idf-svc/pull/484/files#diff-a22c35bd258998dad605171fb28fdca10f6690127e625f0d4fd644f2e246310cR684)

How to implement? In `esp-idf-sys` build script, we can just hard-code an array with all released versions since 4.4.0 (or whatever start), and then use it to generate the cfgs:
```rust
/// They will not be that many, and over time when we drop support for older versions, we'll remove from the list
/// As new versions arrive and we do need to reason about these, we'll add to the list
const VERSIONS: &[(u8, u8, u8)] = &[(u8, u8, u8)] {
    (4, 4, 0), (4, 4, 1), (4, 4, 2), (4, 4, 3), (4, 4, 4), (4, 4, 5), (4, 4, 6), (4, 4, 7),
    (5, 0, 0), (5, 0, 1), ...
    (5, 1, 0), (5, 1, 1), ...
    (5, 2, 0), (5, 2, 1), ...
    (5, 3, 0), ...
};

for (maj, min, patch) in  VERSIONS {
    if esp_idf_version_major > maj 
        || esp_idf_version_major == maj && esp_idf_version_min > min 
        || esp_idf_version_major == maj && esp_idf_version_min == min  && esp_idf_version_patch >= patch
    {
        // print the cfg
        println!("cargo::rustc-cfg=esp_idf_min_version_{maj}_{min}_{patch}");
    }
}

Vollbrecht, 2024-09-18:
Yeah the current cfg's situation gets out of hand on certain parts. Creating separate features for a concrete boundary sounds like a good idea. 

Though we might get into problems, when stuff is back-ported. For example 5.0.2 and 5.1.1 would not be compatible  e.g our minimal version for a thing would be 5.1.2, but they introduced  a 5.0.3 that is also compatible. That kinda logic would still explode. 

Though for this special cases we could still use the current pattern as a backup and replace it for every other place where possible.

ivmarkov, 2024-09-18:
> Yeah the current cfg's situation gets out of hand on certain parts. Creating separate features for a concrete boundary sounds like a good idea.
>
> Though we might get into problems, when stuff is [...]

True. :( 
I have not yet seen such a case w.r.t. API changes, let's hope these are rare.

Though, if you want to express ver >= 5.0.3 && ver < 5.1.0 || ver >= 5.1.2 you can do:
```rust
#[cfg(
    any(
        all(esp_idf_min_version = "5.0.3", not(esp_idf_min_version = "5.1.0")), 
        esp_idf_min_version = "5.1.2",
    )
    ))]
```

... not that readable anymore, yet still more readable than the massacre we have right now.

> 
> Though for this special cases we could still use the current pattern as a backup and replace it for every other place where possible.

Yes, current patterns are not to be retired.

closed 2025-08-12 by ivmarkov

ivmarkov, 2025-08-12:
The list of CI vsrsions for the sys crate:
          - v4.4.8
          - v5.0.9
          - v5.1.6
          - v5.2.5
          - v5.3.3
          - v5.4.2
          - v5.5

A shorter list for the hal/svc crate:
          - v5.1.6?
          - v5.2.5
          - v5.3.3
          - v5.4.2
          - v5.5

For the "next" CI build (which is expected to fail most of the time):
          - master (soon to become v6.0.0 and then master separately)

ivmarkov, 2025-08-12:
BTW, there is an upcoming PR for the hal you need to review. I just pushed the changes to the sys crate as I did not anticipate so many problems. 

The HAL changes are not that many, but better safe than sorry. svc will follow up next.

ivmarkov, 2025-08-12:
Huh. 5.0.9 seems no longer buildable: https://github.com/esp-rs/esp-idf-sys/actions/runs/16909785881/job/47908119094

Shall I indeed, finally retire 4.4.x and 5.0.x from the CI? (Removing the `cfg`s is much trickier and could wait)?