esp-rs/esp-idf-sys #343: esp-idf-sys-0.35.0: error[E0308]: mismatched types

omani, 2024-10-02:
something has broken in the last month or so. this always worked for me. now it doesnt.

First install ESP-IDF, install py_env for esp32c3 and activate py environment:
```
mkdir ~/esp
cd ~/esp
git clone -b v5.2.3 --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh esp32c3
. ~/esp/esp-idf/export.sh
```

now cargo generate a project from the esp-idf-template:
```
sun@ubuntu-embedded:~/shared/git $ cargo generate --git https://github.com/esp-rs/esp-idf-template
✔ 🤷   Which template should be expanded? · cargo
🤷   Project Name: esp-idf-websocket
🔧   Destination: /home/sun/shared/git/esp-idf-websocket ...
🔧   project-name: esp-idf-websocket ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c3
✔ 🤷   Configure advanced template options? · false
🔧   Moving generated files into: `/home/sun/shared/git/esp-idf-websocket`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /home/sun/shared/git/esp-idf-websocket
sun@ubuntu-embedded:~/shared/git $ cd esp-idf-websocket/
```

first of all, the esp-idf-template has the wrong version. it is `v5.2.2` but it should be `v5.2.3`. (esp-idf upstream has released v5.2.3 last month).

so I bump the version in `.cargo/config.toml` to `v5.2.3`:
`ESP_IDF_VERSION = "v5.2.3"`
(does not matter anyway, the error still exists with `v5.2.2`).

then `cargo build`.

fails with:
```
error[E0308]: mismatched types
  --> /home/sun/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:63:62
   |
63 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /home/sun/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:65:51
   |
65 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` (lib) due to 2 previous errors
```

after failed built, check if env vars are set:
```
$ env | rg 'IDF|ESP'
IDF_PYTHON_ENV_PATH=/home/sun/.espressif/python_env/idf5.2_py3.12_env
IDF_PATH=/home/sun/esp/esp-idf
ESP_IDF_VERSION=5.2
IDF_DEACTIVATE_FILE_PATH=/tmp/tmp9gymmg0kidf_19534
IDF_TOOLS_EXPORT_CMD=/home/sun/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/home/sun/esp/esp-idf/install.sh
ESP_ROM_ELF_DIR=/home/sun/.espressif/tools/esp-rom-elfs/20230320/
```
looks good. `ESP_IDF_VERSION=5.2` has been set by `cargo build`. all the others are from activating the ESP-IDF python environment (via `export.sh`).

double check to make sure that the exported ESP-IDF is really being used instead of build script fetching ESP-IDF from source:
```
$ ls -l .embuild
".embuild": No such file or directory (os error 2)
```

double check that `.cargo/config.toml` rustflags look good:
```
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110
```
the template does not include `, "-C", "default-linker-libraries"` (mentioned in previous issues regarding this topic), which - I guess - doesnt matter much here.

I know there were a few issues regarding this problem opened in the past. but I think _this one_ seems to be a bug.

omani, 2024-10-02:
the error occurs also with no preinstalled ESP-IDF and the build script downloads the ESP-IDF repo and installs everything automatically.

Vollbrecht, 2024-10-03:
We do not set any `ENV` variables persistently. They are only injected before the build script runs into the process directly. 
ESP_IDF_VERSION=5.2 is a specific tag that is neither 5.2.2 nor 5.2.3. 

If there are ENV variables in your environment like the IDF_PATH we will use that over a potential cloned repo by the build script. If you are having a IDF_PATH declared in your env you need to make sure that the version exactly matches the version set in the .cargo/config.toml. So is your version located in `IDF_PATH=/home/sun/esp/esp-idf` v5.2.2 ?

Vollbrecht, 2024-10-03:
Can you please copy/paste the build log that is inside `target/riscv.../debug/build/<your-crate-name>.../output ?

yaghmr, 2024-10-03:
I have the same issue, this is my `output` file:

```
cargo:rustc-cfg=esp_idf_adc_cali_efuse_tp_enable
cargo:rustc-cfg=esp_idf_app_build_bootloader
cargo:rustc-cfg=esp_idf_mbedtls_ecp_dp_secp256r1_enabled
cargo:rustc-cfg=esp_idf_mbedtls_tls_server
cargo:rustc-cfg=esp_idf_soc_adc_dma_supported
cargo:rustc-cfg=esp_idf_app_build_use_flash_sections
cargo:rustc-cfg=esp_idf_mqtt_transport_websocket
cargo:rustc-cfg=esp_idf_adc_cal_efuse_vref_enable
cargo:rustc-cfg=esp_idf_soc_spi_mem_support_config_gpio_by_efuse
cargo:rustc-cfg=esp_idf_esptoolpy_flashfreq_40m
cargo:rustc-cfg=esp_idf_rtc_clk_src_int_rc
cargo:rustc-cfg=esp_idf_periph_ctrl_func_in_iram
cargo:rustc-cfg=esp_idf_spi_flash_vendor_gd_supported
cargo:rustc-cfg=esp_idf_lwip_tcp_queue_ooseq
cargo:rustc-cfg=esp_idf_soc_i2s_supports_lcd_camera
cargo:rustc-cfg=esp_idf_soc_mpu_supported
cargo:rustc-cfg=esp_idf_lwip_esp_mldv6_report
cargo:rustc-cfg=esp_idf_soc_xtal_support_26m
cargo:rustc-cfg=esp_idf_mbedtls_internal_mem_alloc
cargo:rustc-cfg=esp_idf_log_maximum_equals_default
cargo:rustc-cfg=esp_idf_spi_flash_brownout_reset_xmc
cargo:rustc-cfg=esp_idf_esp_wifi_dynamic_tx_buffer
cargo:rustc-cfg=esp_idf_mbedtls_pem_write_c
cargo:rustc-cfg=esp_idf_soc_rtc_slow_mem_supported
cargo:rustc-cfg=esp_idf_soc_spi_as_cs_supported
cargo:rustc-cfg=esp_idf_mbedtls_ecp_dp_bp256r1_enabled
cargo:rustc-cfg=esp_idf_soc_sdmmc_host_supported
cargo:rustc-cfg=esp_idf_spi_flash_vendor_mxic_supported
cargo:rustc-cfg=esp_idf_soc_clk_tree_supported
cargo:rustc-cfg=esp_idf_soc_configurable_vddsdio_supported
cargo:rustc-cfg=esp_idf_soc_pm_support_ext_wakeup
cargo:rustc-cfg=esp_idf_spiffs_page_check
cargo:rustc-cfg=esp_idf_mbedtls_ssl_alpn
cargo:rustc-cfg=esp_idf_mbedtls_ssl_proto_tls1_2
cargo:rustc-cfg=esp_idf_soc_secure_boot_supported
cargo:rustc-cfg=esp_idf_wifi_prov_sta_all_channel_scan
cargo:rustc-cfg=esp_idf_idf_target_arch_xtensa
cargo:rustc-cfg=esp_idf_esp_timer_task_affinity_cpu0
cargo:rustc-cfg=esp_idf_esp_timer_isr_affinity_cpu0
cargo:rustc-cfg=esp_idf_compiler_hide_paths_macros
cargo:rustc-cfg=esp_idf_mbedtls_x509_crl_parse_c
cargo:rustc-cfg=esp_idf_esp_tls_using_mbedtls
cargo:rustc-cfg=esp_idf_esp_wifi_ampdu_tx_enabled
cargo:rustc-cfg=esp_idf_esp_event_post_from_iram_isr
cargo:rustc-cfg=esp_idf_soc_mcpwm_supported
cargo:rustc-cfg=esp_idf_soc_pm_support_rc_fast_pd
cargo:rustc-cfg=esp_idf_adc_disable_dac_output
cargo:rustc-cfg=esp_idf_soc_ulp_has_adc
cargo:rustc-cfg=esp_idf_soc_gpspi_supported
cargo:rustc-cfg=esp_idf_mbedtls_key_exchange_rsa
cargo:rustc-cfg=esp_idf_spi_flash_brownout_reset
cargo:rustc-cfg=esp_idf_soc_pm_support_ext1_wakeup
cargo:rustc-cfg=esp_idf_mbedtls_gcm_c
cargo:rustc-cfg=esp_idf_twai_errata_fix_tx_intr_lost
cargo:rustc-cfg=esp_idf_soc_dac_supported
cargo:rustc-cfg=esp_idf_esp_protocomm_support_security_version_1
cargo:rustc-cfg=esp_idf_partition_table_single_app
cargo:rustc-cfg=esp_idf_soc_rtc_mem_supported
cargo:rustc-cfg=esp_idf_esp_task_wdt_init
cargo:rustc-cfg=esp_idf_lwip_tcp_oversize_mss
cargo:rustc-cfg=esp_idf_esp_ipc_isr_enable
cargo:rustc-cfg=esp_idf_hal_spi_slave_func_in_iram
cargo:rustc-cfg=esp_idf_mbedtls_certificate_bundle_default_full
cargo:rustc-cfg=esp_idf_spi_master_isr_in_iram
cargo:rustc-cfg=esp_idf_apptrace_dest_uart_none
cargo:rustc-cfg=esp_idf_wl_sector_size_4096
cargo:rustc-cfg=esp_idf_esp_time_funcs_use_esp_timer
[... 408 lines cut ...]
cargo:rustc-cfg=esp_idf_comp_esp_eth_enabled
cargo:rustc-cfg=esp_idf_comp_main_enabled
cargo:rustc-cfg=esp_idf_comp_app_trace_enabled
cargo:rustc-cfg=esp_idf_comp_xtensa_enabled
cargo:rustc-cfg=esp_idf_comp_efuse_enabled
cargo:rustc-cfg=esp_idf_comp_esp_partition_enabled
cargo:rustc-cfg=esp_idf_comp_esp_system_enabled
cargo:rustc-cfg=esp_idf_comp_hal_enabled
cargo:rustc-cfg=esp_idf_comp_nvs_flash_enabled
cargo:rustc-cfg=esp_idf_comp_ieee802154_enabled
cargo:rustc-cfg=esp_idf_comp_esp_http_server_enabled
cargo:rustc-cfg=esp_idf_comp_wifi_provisioning_enabled
cargo:rustc-cfg=esp_idf_comp_tcp_transport_enabled
cargo:rustc-cfg=esp_idf_comp_usb_enabled
cargo:rustc-cfg=esp_idf_comp_newlib_enabled
cargo:rustc-cfg=esp_idf_comp_wpa_supplicant_enabled
cargo:rustc-cfg=esp_idf_comp_bt_enabled
cargo:rustc-cfg=esp_idf_comp_espcoredump_enabled
cargo:rustc-cfg=esp_idf_comp_mqtt_enabled
cargo:rustc-cfg=esp_idf_comp_esp_bootloader_format_enabled
cargo:rustc-cfg=esp_idf_comp_heap_enabled
cargo:rustc-cfg=esp_idf_comp_pthread_enabled
cargo:rustc-cfg=esp_idf_comp_esp_phy_enabled
cargo:rustc-cfg=esp32
```

yaghmr, 2024-10-03:
Adding these flags worked for me:

```
[build]
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]
```

ivmarkov, 2024-10-03:
> Adding these flags worked for me:
> 
> ```
> [build]
> rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]
> ```

These flags should _anyway_ be in your `.cargo/config.toml` file if your binary crate project was configured correctly. One way to do this is by using the `esp-idf-template` project and carefully following all steps.

> I have the same issue, this is my output file: cargo:rustc-cfg=esp_idf_adc_cali_efuse_tp_enable cargo:rustc-cfg=esp_idf_app_build_bootloader cargo:rustc-cfg=esp

This ^^^ you should absolutely not do. The way to configure ESP IDF, even when Rust is used is `sdkconfig.defaults`. You might also want to check out the Rust ESP-RS book.

omani, 2024-10-03:
ok so I have these two containers I do my rust things in (LXD containers). one container always worked, so I took it as a base container (cloned it) to see if I can reproduce this problem.

after many hours of trying I think I finally figured out what the problem is.

what I did was a system update (sudo apt update && sudo apt upgrade). and I remember that was the only thing I did on this "new" container before it started to fail. so I cloned again from my last working snapshot and took a look what actually is changing on the system. and it turns out it is the python3 package.

I tried it again. clone from working container where same steps from above work (cargo generate, cargo build, etc.) and try it again after a system update, and it fails.

Im pretty sure it is because of this, because there was one thing that gave it away.

I had exported the ESP-IDF environment before doing the system update (and a successful cargo build) with `. ~/esp/esp-idf/export.sh`.

I did the system update and ran `cargo build` again and it worked. then I thought "wait a minute. python has changed, but my py environment is still the old one (due to the export.sh). so I spawned a new shell and did a `. ~/esp/esp-idf/export.sh` again. and then `cargo build` fails with the above `mismatched types` error.

so I am 100% sure it is due to the new python version.

omani, 2024-10-03:
> So is your version located in IDF_PATH=/home/sun/esp/esp-idf v5.2.2 ?
yes. to get the idf version, activate the ESP-IDF environment and check the version with idf.py:
```
$ . ~/esp/esp-idf/export.sh
$ idf.py --version
ESP-IDF v5.4-dev-2306-gdbce23f8a4
```

it is v5.4 now because I switched to a different version. in my tests it was v5.2.3 before. in this case,
btw, what would one say in `ESP_IDF_VERSION`. `v5.4` or `5.4` or exactly the above string?

omani, 2024-10-03:
btw my `ESP_IDF_VERSION` is `ESP_IDF_VERSION = "5.4"` the whole time in my tests. while the `idf.py --version` reports `ESP-IDF v5.4-dev-2306-gdbce23f8a4`. so I guess it does not matter if the version string is exactly the same.

omani, 2024-10-03:
ok here the reproducing of the issue:
working snapshot:
```
$ . ~/esp/esp-idf/export.sh
Setting IDF_PATH to '/home/sun/esp/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Not using an unsupported version of tool xtensa-esp-elf found in PATH: esp-13.2.0_20230928. To use it, run '/usr/bin/python3 /home/sun/esp/esp-idf/tools/idf_tools.py export --prefer-system'
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /home/sun/.espressif/espidf.constraints.v5.4.txt
Requirement files:
 - /home/sun/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/sun/.espressif/python_env/idf5.4_py3.12_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /home/sun/esp/esp-idf/components/espcoredump
  /home/sun/esp/esp-idf/components/partition_table
  /home/sun/esp/esp-idf/components/app_update
  /home/sun/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin
  /home/sun/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin
  /home/sun/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin
  /home/sun/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin
  /home/sun/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin
  /home/sun/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240726/openocd-esp32/bin
  /home/sun/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin
  /home/sun/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin
  /home/sun/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin
  /home/sun/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin
  /home/sun/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin
  /home/sun/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240726/openocd-esp32/bin
  /home/sun/.espressif/python_env/idf5.4_py3.12_env/bin
  /home/sun/esp/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

```

cargo build runs fine:
```
$ cargo clean
     Removed 2994 files, 1.5GiB total
$ cargo build
   Compiling compiler_builtins v0.1.123
   Compiling core v0.0.0 (/home/sun/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.158
   Compiling proc-macro2 v1.0.86
   Compiling unicode-ident v1.0.13
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/sun/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
   Compiling quote v1.0.37
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling syn v2.0.79
   Compiling libc v0.2.159
   Compiling serde v1.0.210
   Compiling bitflags v2.6.0
   Compiling aho-corasick v1.1.3
   Compiling serde_derive v1.0.210
   Compiling crossbeam-utils v0.8.20
   Compiling regex-syntax v0.8.5
   Compiling glob v0.3.1
   Compiling rustc-std-workspace-core v1.99.0 (/home/sun/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling rustix v0.38.37
[... 1105 lines cut ...]
error[E0308]: mismatched types
  --> /home/sun/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:63:62
   |
63 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /home/sun/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:65:51
   |
65 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` (lib) due to 2 previous errors
```

there you go. now it fails!

so I take back what I said. it's not because of some system packages, nor python3, nor libc. it is because of a single line that had been exported without notice and I didnt catch that.

I will remove that line. as it turns out it is somehow dangerous to export `RUSTFLAGS` globally.

problem solved for me.

this was a great adventure of error diagnosis.

ivmarkov, 2024-10-03:
Exporting `RUSTFLAGS` globally - as you do - is a really bad idea. It overwrites everything we set in ./cargo/config.toml.

closed 2024-10-03 by ivmarkov