esp-rs/esp-idf-sys #345: IDF_TARGET not set

hhy50, 2024-10-24:
After adding this parameter to Cargo.toml, the build failed
```toml
[lib]
proc-macro = true
```

build log:
```txt
--- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'x86_64-apple-darwin'
warning: build failed, waiting for other jobs to finish...
```

ivmarkov, 2024-10-24:
What do you expect? You are declaring a proc-maceo library. These run on the host.

closed 2024-10-24 by ivmarkov

hhy50, 2024-10-24:
```toml
[[bin]]
name = "billiards_esp32"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[lib]
proc-macro = true
```
but, I also have a bin crate

ivmarkov, 2024-10-24:
It does not work like that. You are chasing rust restrictions, misinterpreting the error and then opening as bug here. What `IDG_TARGET`? Look at the error output. 

> Error: Unsupported target 'x86_64-apple-darwin'

It is rustc and cargo themselves who are not picking your esp32 target, not `esp-idf-sys`. You are compiling **for the host**.

hhy50, 2024-10-24:
> It does not work like that. You are chasing rust restrictions, misinterpreting the error and then opening as bug here. What `IDG_TARGET`? Look at the error output.
>
>
> It is rustc and cargo themselves [...]

```toml
[lib]
proc-macro = true
```
Why can he run it after deleting this

hhy50, 2024-10-24:
@ivmarkov 
You can find an example project and try adding it

ivmarkov, 2024-10-24:
OK I give it one last try: what you are trying to do (mixing a proc-macro crate together in another, non-proc-macro crate) is just **NOT** possible in Rust.

This is not a problem of `esp-idf-sys`. 
This is a restriction of Rust. And if you think about it, it does make sense!

Try to do what you are trying to do with any other embedded framework - be it `esp-hal` or `embassy` and you'll see it is simply impossible.

The **only** solution is to have your proc-macros in a separate, **dedicated** proc-macro crate. As everybody else does. Google the internet for that.