esp-rs/esp-idf-sys #347: can't compile the esp-idf sys

phantomboy0, 2024-10-24:
im trying to build it for my ESP32 and i get the following result , ask me if any more info is needed:
 i got cl.exe and gcc minwg-w64 installed and defined in PATH and they work fine in powershell
 
 `# cargo build --release --verbose
...
  Fresh embassy-time-driver v0.1.0
       Fresh embedded-svc v0.28.0
       Fresh embassy-futures v0.1.1
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `R:\e\target\release\build\esp-idf-sys-7c6c191c1ff87c20\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.12.7
  pip 24.2 from R:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\Lib\site-packages\pip (python 3.12)
  Requirement already satisfied: pip in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (24.2)
  Requirement already satisfied: setuptools in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-75.2.0-py3-none-any.whl.metadata (6.9 kB)
  Using cached setuptools-75.2.0-py3-none-any.whl (1.2 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
      Found existing installation: setuptools 71.0.0
      Uninstalling setuptools-71.0.0:
        Successfully uninstalled setuptools-71.0.0
  Successfully installed setuptools-75.2.0
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Collecting setuptools (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 3))
    Using cached https://dl.espressif.com/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
  Requirement already satisfied: packaging in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 4)) (24.1)
  Requirement already satisfied: click in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 7)) (8.1.7)
  Requirement already satisfied: pyserial in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 8)) (3.5)
  Requirement already satisfied: cryptography in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 9)) (41.0.7)
  Requirement already satisfied: pyparsing in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 10)) (3.2.0)
  Requirement already satisfied: pyelftools in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 11)) (0.31)
  Requirement already satisfied: idf-component-manager in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.0.4)
  Requirement already satisfied: esp-coredump in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 13)) (1.12.0)
  Requirement already satisfied: esptool in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 14)) (4.7.0)
  Requirement already satisfied: esp-idf-kconfig in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 15)) (1.4.3)
  Requirement already satisfied: esp-idf-monitor in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 16)) (1.5.0)
  Requirement already satisfied: esp-idf-size in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 17)) (1.6.0)
  Requirement already satisfied: esp-idf-panic-decoder in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 18)) (1.2.1)
  Requirement already satisfied: pyclang in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 19)) (0.5.0)
  Requirement already satisfied: construct in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 20)) (2.10.70)
  Requirement already satisfied: freertos_gdb in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 23)) (1.0.3)
  Requirement already satisfied: colorama in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from click->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 7)) (0.4.6)
  Requirement already satisfied: cffi>=1.12 in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from cryptography->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 9)) (1.17.1)
  Requirement already satisfied: pyyaml in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from idf-component-manager->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (6.0.2)
  Requirement already satisfied: requests in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from idf-component-manager->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.32.3)
  Requirement already satisfied: requests-file in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from idf-component-manager->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.1.0)
  Requirement already satisfied: requests-toolbelt in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from idf-component-manager->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (1.0.0)
  Requirement already satisfied: tqdm in r:\e\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from idf-component-manager->-r R:\e\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (4.66.5)
[... 264 lines cut ...]
      --recursive' in ESP-IDF directory to fix this.

    Call Stack (most recent call first):

      R:/e/.embuild/espressif/esp-idf/v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
      R:/e/.embuild/espressif/esp-idf/v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)

[... cut ...]
  thread 'main' panicked at C:\Users\Amir Hosain\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cmake-0.1.51\src/lib.rs:1100:5:

  command did not execute successfully, got: exit code: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
 Amir0 on  R:/e master
# cl
Microsoft (R) C/C++ Optimizing Compiler Version 19.41.34120 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

usage: cl [ option... ] filename... [ /link linkoption... ]
`

ivmarkov, 2024-10-24:
> im trying to build it for my ESP32 and i get the following result , ask me if any more info is needed: i got cl.exe and gcc minwg-w64 installed and defined in PATH and they work fine in powershell

These tools ^^^ are not necessary at all and won't help at all.
- Please carefully follow all installation instructions [here](https://github.com/esp-rs/esp-idf-template?tab=readme-ov-file#prerequisites).
- Ideally, first try to build a template as generated by the `esp-idf-template` and then go to building `esp-idf-sys` (which I'm unsure why you need per-se. The interesting stuff (examples) are anyway in `esp-idf-hal` and `esp-idf-svc`).

phantomboy0, 2024-10-24:
@ivmarkov , thanks , I'm trying to write a STD rust and also i'm totall newbie but i got some experience with no_std.

so after running `cargo generate esp-rs/esp-idf-template cargo` as the instruction said , and touching nothing and just building the hello template i get the error again .

it still wants to build esp-idf-sys

```
error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `R:\g\h\target\release\build\esp-idf-sys-d5553dc1b2a013a4\build-script-build` (exit code: 1)
```

ivmarkov, 2024-10-24:
Is "r" a real drive or a "subst" trick?
- if a real drive, remove the ".embuild" dir I side

Vollbrecht, 2024-10-24:
can you run `cargo b -vv | findstr "[esp-idf-sys 0.35.0]`  for a more complete output / concise output of the build failure.

Vollbrecht, 2024-10-24:
from powershell you might use something like "Select-String" instead of "findstr" ?  i only use grep and linux so not to sure whats the correct way to filter is on windows

phantomboy0, 2024-10-24:
@ivmarkov it's a real drive and real dir , but didn't work

@Vollbrecht i got thi out put after `cargo build -vv | Select-String "\[esp-idf-sys 0.35.0\]" | Out-File -FilePath "output.txt"` :

```

[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=MCU
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_VERSION
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=IDF_PATH
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=EXTRA-COMPONENTS
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
[esp-idf-sys 0.35.0] cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
[esp-idf-sys 0.35.0] Python 3.12.7
[esp-idf-sys 0.35.0] pip 24.2 from R:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\Lib\site-packages\pip (python 3.12)
[esp-idf-sys 0.35.0] Requirement already satisfied: pip in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (24.2)
[esp-idf-sys 0.35.0] Requirement already satisfied: setuptools in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (71.0.0)
[esp-idf-sys 0.35.0] Collecting setuptools
[esp-idf-sys 0.35.0]   Using cached setuptools-75.2.0-py3-none-any.whl.metadata (6.9 kB)
[esp-idf-sys 0.35.0] Using cached setuptools-75.2.0-py3-none-any.whl (1.2 MB)
[esp-idf-sys 0.35.0] Installing collected packages: setuptools
[esp-idf-sys 0.35.0]   Attempting uninstall: setuptools
[esp-idf-sys 0.35.0]     Found existing installation: setuptools 71.0.0
[esp-idf-sys 0.35.0]     Uninstalling setuptools-71.0.0:
[esp-idf-sys 0.35.0]       Successfully uninstalled setuptools-71.0.0
[esp-idf-sys 0.35.0] Successfully installed setuptools-75.2.0
[esp-idf-sys 0.35.0] Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
[esp-idf-sys 0.35.0] Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
[esp-idf-sys 0.35.0] Collecting setuptools (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 3))
[esp-idf-sys 0.35.0]   Using cached https://dl.espressif.com/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
[esp-idf-sys 0.35.0] Requirement already satisfied: packaging in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 4)) (24.1)
[esp-idf-sys 0.35.0] Requirement already satisfied: click in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 7)) (8.1.7)
[esp-idf-sys 0.35.0] Requirement already satisfied: pyserial in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 8)) (3.5)
[esp-idf-sys 0.35.0] Requirement already satisfied: cryptography in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 9)) (41.0.7)
[esp-idf-sys 0.35.0] Requirement already satisfied: pyparsing in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 10)) (3.2.0)
[esp-idf-sys 0.35.0] Requirement already satisfied: pyelftools in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 11)) (0.31)
[esp-idf-sys 0.35.0] Requirement already satisfied: idf-component-manager in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.0.4)
[esp-idf-sys 0.35.0] Requirement already satisfied: esp-coredump in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 13)) (1.12.0)
[esp-idf-sys 0.35.0] Requirement already satisfied: esptool in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 14)) (4.7.0)
[esp-idf-sys 0.35.0] Requirement already satisfied: esp-idf-kconfig in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 15)) (1.4.3)
[esp-idf-sys 0.35.0] Requirement already satisfied: esp-idf-monitor in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 16)) (1.5.0)
[esp-idf-sys 0.35.0] Requirement already satisfied: esp-idf-size in r:\g\h\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (from -r R:\g\h\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 17)) [... cut ...]
[esp-idf-sys 0.35.0] Bootloader binary size 0x6860 bytes. 0x7a0 bytes (7%) free.
[esp-idf-sys 0.35.0] cargo:root=R:\g\h\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-90d18ba0c7b95406\out
[esp-idf-sys 0.35.0] cargo:rerun-if-changed=C:\Users\Amir Hosain\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-idf-sys-0.35.0\src\include\esp-idf\bindings.h
[esp-idf-sys 0.35.0] cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=R:\g\h\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-90d18ba0c7b95406\out\bindings.rs

 ```

phantomboy0, 2024-10-24:
@Vollbrecht also this is the full output:
[PowerShell.txt](https://github.com/user-attachments/files/17507805/PowerShell.txt)

Vollbrecht, 2024-10-24:
hmm might be [this](https://github.com/esp-rs/espup/issues/440)  problem. Can you check that out please.

phantomboy0, 2024-10-24:
i dont have eps tho

> I just ran esp uninstall and esp install --toolchain-version 1.77.0 to install an older toolchain, then I got complied successfully. 

> Maybe there is something wrong with the newer veriosn?

i dont have esp command only espup

phantomboy0, 2024-10-24:
@Vollbrecht https://github.com/esp-rs/espup/issues/440#issuecomment-2250682346

how i can install espup 0.11.0 instead of 0.12.2 for rust 1.81?

phantomboy0, 2024-10-24:
i can confirm replacing 
llvm-esp-16.0.4-20231113-win64.tar.xz/esp-clang [https://github.com/espressif/llvm-project/releases/tag/esp-16.0.4-20231113](From Here) inside the `C:\Users\Amir Hosain\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang` worked for me and i did a `cargo clean ` and then build

closed 2024-10-24 by phantomboy0