esp-rs/esp-idf-sys #348: build fail

hhy50, 2024-11-06:
cargo.toml:
```toml
[package]
name = "demo"
version = "0.1.0"
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "demo"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = []

experimental = ["esp-idf-svc/experimental"]

[dependencies]
log = "0.4"
esp-idf-svc = { version = "0.49", features = ["critical-section", "embassy-time-driver", "embassy-sync"] }

[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
[build-dependencies]
embuild = "0.32.0"
cc = "=1.1.30" # Necessary until a new version of `esp-idf-sys` is released

```

log:
```
   Compiling esp-idf-svc v0.49.1 (https://github.com/esp-rs/esp-idf-svc#e1d0f285)
error[E0599]: no method named `no_default_flags` found for mutable reference `&mut Config` in the current scope
   --> /home/hanhaiyang/.cargo/git/checkouts/esp-idf-sys-d4c2e811b42d99d6/e8ef996/build/native/cargo_driver.rs:461:10
    |
457 | /     cmake_config
458 | |         .generator(cmake_generator.name())
459 | |         .out_dir(&out_dir)
460 | |         .no_build_target(true)
461 | |         .no_default_flags(true)
    | |         -^^^^^^^^^^^^^^^^ method not found in `&mut Config`
    | |_________|
    |

```

leaderiop, 2024-11-08:
same probleme here

hhy50, 2024-11-08:
@leaderiop You can try this, I haven't had a chance to try it yet [hardware-check build fails with "unknown target riscv32imc_zicsr_zifencei-esp-espidf](https://github.com/esp-rs/std-training/issues/266)

closed 2024-11-08 by ivmarkov

hhy50, 2024-11-09:
@ivmarkov hi, Why does this error occur when I delete the patch for esp-idf-sys?

ivmarkov, 2024-11-09:
> @ivmarkov hi, Why does this error occur when I delete the patch for esp-idf-sys?
> 
> ```
>   Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2") }
>   Using esp-idf v5.2.0 at '/home/hanhaiyang/.espressif/esp-idf/v5.2'
> 
>   error occurred: unknown target `riscv32imc_zicsr_zifencei-esp-espidf`
> ```

Because the newly released cc-rs crate does not support this trick anymore.
There is a long discussion about it in the chat room if you are interested, search there for `zicsr`