esp-rs/esp-idf-sys #349: Failure to add remote component

indexds, 2024-11-10:
Hello,

Even though the documentation is quite extensive, I still fail to add a remote component to my project.

I have read [this](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#remote-components-idf-component-registry) and it states that to add a remote component to my cargo project I should do something like this: 

```rust
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "espressif/mdns", version = "1.4" }

[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "trombik/esp_wireguard", version = "0.9" }
bindings_module = "wireguard"
```
Now, mdns is working just fine, most likely because it already exists in [bindings.h](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h)
But what about this wireguard module? It does not exist in the bindings.h, and I can't just specify a header path as it is generated at compile time with a bunch of random characters in the target folder like this:
```
target\xtensa-esp32-espidf\release\build\esp-idf-sys-2420579a46d33aef\out\managed_components\trombik__esp_wireguard\include\esp_wireguard.h
```

Ideally, I would like to do something like this:

```rust
use esp_idf_svc::sys::wireguard::{
    esp_wireguard_connect,
    esp_wireguard_init,
    // esp_wireguard_disconnect,
    // esp_wireguard_set_default,
    esp_wireguardif_peer_is_up,
    wireguard_config_t,
    wireguard_ctx_t,
};
```
And have access to all my C functions. Currently, the program fails at compile time because it cannot find the wireguard module inside sys.

Where am I going wrong?
Thanks in advance.

ivmarkov, 2024-11-10:
You are missing the `bindings_header` parameter. Something like this should work:
```toml
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "trombik/esp_wireguard", version = "0.9" }
bindings_header = "esp_wireguard.h"
```

With that said, the component you are trying to use hasn't been updated since forever, and it does not even list ESP IDF 5.X as compatible.
And in the meantime, ESP IDF 4.4.X is out of support (even if we still build for it).

ivmarkov, 2024-11-10:
> Hi, and yeah I tried doing that already, but it looks for the header in the root of the project, where the Cargo.toml file is located.

Yes, I've missed a step, sorry. Do this instead:
```toml
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "trombik/esp_wireguard", version = "0.9" }
bindings_header = "esp_wireguard_bindings.h"
```

Then, create an `esp_wireguard_bindings.h` file in the root of your crate (next to `Cargo.toml`), and inside it just do:
```c
#pragma once

#include "esp_wireguard.h"
```

You can also move `esp_wireguard_bindings.h` inside `src` or wherever, but then you need to prefix it for the `bindings_header` thing, i.e. `src/esp_wireguard_bindings.h`.

>  As for the component itself, I haven't found anything more recent, and figured I'd try anyway as I don't think there has to be anything version specific inside, though I may be wrong.

I'm not holding my breath, but sure, you could try.

ivmarkov, 2024-11-10:
If you are wondering, the extra level of indirection (hosting your own bindings header) is necessary in case the local or remote component you are trying to use has multiple headers, and you need to include all of them.

indexds, 2024-11-10:
Okay, that fixed it thank you very much!

> If you are wondering, the extra level of indirection (hosting your own bindings header) is necessary in case the local or remote component you are trying to use has multiple headers, and you need to include all of them.

That makes a ton of sense, though I didn't seem to think so 3 hours ago because I blatantly ignored creating the extra header file thinking it was stupid. 

Since I'm trying to make this a learning experience, how does the "esp_wireguard_bindings.h" file even see "esp_wireguard.h" ? What's the glue?

indexds, 2024-11-10:
Also, the intellisense can't see the module and the exposed functions inside, is there a way to fix this or am I doomed to make spelling mistakes?

indexds, 2024-11-10:
> Also, the intellisense can't see the module and the exposed functions inside, is there a way to fix this or am I doomed to make spelling mistakes?

Nevermind I fixed my own issue by rebuilding rust-analyzer's build dependencies here: 

[image]

ivmarkov, 2024-11-10:
> Since I'm trying to make this a learning experience, how does the "esp_wireguard_bindings.h" file even see "esp_wireguard.h" ? What's the glue?

It gets it from cmake which reports all include dirs.

closed 2024-11-11 by indexds