esp-rs/esp-idf-sys #352: phy_init crash with basic wifi snippet

indexds, 2024-12-01:
Hello, 

I've been struggling for the past few hours with this. Running a simple wifi snippet as shown below results in a crash with no backtrace when wifi.start() is called.

```rust
fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().expect("Failed to take peripherals!");
    let sysloop = EspSystemEventLoop::take().expect("Failed to take sysloop!");
    let nvs = EspDefaultNvsPartition::take().ok();

    let mut wifi = WifiDriver::new(peripherals.modem, sysloop.clone(), nvs)
    .expect("Failed to init WifiDriver!");

    let wifi_config = Configuration::Client(ClientConfiguration {
        ssid: "foo".try_into().unwrap(),
        auth_method: AuthMethod::WPA2Personal,
        password: "bar".try_into().unwrap(),
        ..Default::default()
    });

    wifi.set_configuration(&wifi_config).expect("failed to set config");
    wifi.start().expect("failed to start driver");
}
```

```log
I (60) boot: ESP-IDF v5.2.3 2nd stage bootloader
I (60) boot: compile time Dec  1 2024 17:17:56
I (60) boot: Multicore bootloader
I (65) boot: chip revision: v1.1
I (69) qio_mode: Enabling default flash chip QIO
I (74) boot.esp32: SPI Speed      : 80MHz
I (79) boot.esp32: SPI Mode       : QIO
I (83) boot.esp32: SPI Flash Size : 4MB
I (88) boot: Enabling RNG early entropy source...
I (93) boot: Partition Table:
I (97) boot: ## Label            Usage          Type ST Offset   Length
I (104) boot:  0 nvs              WiFi data        01 02 00012000 00004000
I (112) boot:  1 nvs_keys         NVS keys         01 04 00017000 00001000
I (119) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (127) boot:  3 factory          factory app      00 00 00020000 00200000
I (134) boot: End of partition table
I (139) boot: Defaulting to factory image
I (143) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=23534h (144692) map
I (189) esp_image: segment 1: paddr=0004355c vaddr=3ffb0000 size=03d48h ( 15688) load
I (194) esp_image: segment 2: paddr=000472ac vaddr=40080000 size=08d6ch ( 36204) load
I (206) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=9c8bch (641212) map
I (370) esp_image: segment 4: paddr=000ec8e4 vaddr=40088d6c size=0e644h ( 58948) load
I (400) boot: Loaded app from partition at offset 0x20000
I (400) boot: Disabling RNG early entropy source...
I (412) cpu_start: Multicore app
I (421) cpu_start: Pro cpu start user code
I (421) cpu_start: cpu freq: 160000000 Hz
I (421) cpu_start: Application information:
I (424) cpu_start: Project name:     libespidf
I (429) cpu_start: App version:      6c6ad98-dirty
I (434) cpu_start: Compile time:     Dec  1 2024 17:17:31
I (440) cpu_start: ELF file SHA256:  000000000...
I (446) cpu_start: ESP-IDF:          v5.2.3
I (451) cpu_start: Min chip rev:     v0.0
I (455) cpu_start: Max chip rev:     v3.99 
I (460) cpu_start: Chip rev:         v1.1
I (465) heap_init: Initializing. RAM available for dynamic allocation:
I (472) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (478) heap_init: At 3FFB74C8 len 00028B38 (162 KiB): DRAM        
I (484) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM       
I (491) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM      
I (497) heap_init: At 400973B0 len 00008C50 (35 KiB): IRAM
I (504) spi_flash: detected chip: generic
I (508) spi_flash: flash io: qio
W (512) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (521) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (530) main_task: Started on CPU0
I (540) main_task: Calling app_main()
I (570) wifi:wifi driver task: 3ffbdf9c, prio:23, stack:6656, core=0
I (580) wifi:wifi firmware version: 82f54d2
I (580) wifi:wifi certification version: v7.0
I (580) wifi:config NVS flash: enabled
I (580) wifi:config nano formating: disabled
I (580) wifi:Init data frame dynamic rx buffer num: 32
I (590) wifi:Init static rx mgmt buffer num: 5
I (590) wifi:Init management short buffer num: 32
I (590) wifi:Init dynamic tx buffer num: 32
I (600) wifi:Init static rx buffer size: 1600
I (600) wifi:Init static rx buffer num: 10
I (610) wifi:Init dynamic rx buffer num: 32
I (610) wifi_init: rx ba win: 6
I (610) wifi_init: tcpip mbox: 32
I (620) wifi_init: udp mbox: 6
I (620) wifi_init: tcp mbox: 6
I (630) wifi_init: tcp tx win: 5760
I (630) wifi_init: tcp rx win: 5760
I (630) wifi_init: tcp mss: 1440
I (640) wifi_init: WiFi IRAM OP enabled
I (640) wifi_init: WiFi RX IRAM OP enabled
I (650) phy_init: phy_version 4840,a40f0ac,Aug 16 2024,16:48:06    
W (650) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
�ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

Everything in sdkconfig is set to the defaults, so I struggle to see why this would crash... It's most likely due to phy_init but beyond that I have no clue. 

Any ideas?

ivmarkov, 2024-12-01:
Brownout (not enough power) via USB?

indexds, 2024-12-01:
Holy shit that was it. Many thanks. I was looking for a software issue, no wonder..

For anyone ever coming here, the issue was that starting wifi takes a lot of power and I was supplying the board with 3.3V. Switching to 5V fixed the issue.

closed 2024-12-01 by indexds

ivmarkov, 2024-12-01:
> Holy shit that was it. Many thanks. I was looking for a software issue, no wonder..
> 
> For anyone ever coming here, the issue was that starting wifi takes a lot of power and I was supplying the board with 3.3V. Switching to 5V fixed the issue.

Keep in mind that supplying the board with 5v should only be done on the 5v pin, or else you'll burn it!!

The chip itself works on 3.3v.
The 5v pin is connected to a step-down converter to 3.3v

Also you are missing amperes not voltage...