esp-rs/esp-idf-sys #354: ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t; under Linux/WSL2 Ubuntu 22.04 for Esp32

emeric-martineau, 2024-12-14:
Hello,

I have update my Rust tool chain with the last version of Rust for ESP32:
```
$ espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[info]: Downloading 'rust.tar.xz'
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[info]: Creating symlink between '/home/xxx/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/home/xxx/.espup/esp-clang'
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Downloading 'rust-src.tar.xz'
[info]: Installing 'rust-src' component for Xtensa Rust toolchain
[info]: Installation successfully completed!
```

Now, when I compile my project, I have this error:
```
error[E0308]: mismatched types
  --> /home/xxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:65:51
   |
65 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i64`, found `i32`

In .cargo/config.toml file, I had rustflags = ["--cfg", "espidf_time64"], but without success. I have this error:

error[E0308]: mismatched types
  --> /home/xxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:63:62
   |
63 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i64`, found `i32`

error[E0308]: mismatched types
  --> /home/xxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/lib.rs:65:51
   |
65 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i64`, found `i32`
```

I try to use ESP-IDF v5.3.2 (previously `v4.4.6`) but I got this error:
```
   Compiling embedded-svc v0.28.0
error[E0560]: struct `esp_idf_hal::sys::eth_esp32_emac_config_t` has no field named `smi_mdc_gpio_num`
   --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/eth.rs:317:13
    |
317 |             smi_mdc_gpio_num: mdc,
    |             ^^^^^^^^^^^^^^^^ `esp_idf_hal::sys::eth_esp32_emac_config_t` does not have this field
    |
    = note: available fields are: `__bindgen_anon_1`, `clock_config`, `dma_burst_len`, `intr_priority`

error[E0560]: struct `esp_idf_hal::sys::eth_esp32_emac_config_t` has no field named `smi_mdio_gpio_num`
   --> /home/xxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/eth.rs:318:13
    |
318 |             smi_mdio_gpio_num: mdio,
    |             ^^^^^^^^^^^^^^^^^ `esp_idf_hal::sys::eth_esp32_emac_config_t` does not have this field
    |
    = note: available fields are: `__bindgen_anon_1`, `clock_config`, `dma_burst_len`, `intr_priority`
```

My `.cargo/config.toml` file:
```
[build]
target = ["xtensa-esp32-espidf"]

[target.xtensa-esp32-espidf]
linker = "ldproxy"
rustflags = ["--cfg", "espidf_time64"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
# Note: these variables are not used when using pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v4.4.6"
```

The build.rs file:
```
// Necessary because of this issue: https://github.com/rust-lang/cargo/issues/9641
fn main() -> Result<(), Box<dyn std::error::Error>> {
    embuild::build::CfgArgs::output_propagated("ESP_IDF")?;
    embuild::build::LinkArgs::output_propagated("ESP_IDF")?;
    Ok(())
}
```

The `Cargo.toml` file:
```
[package]
...
resolver = "2"
rust-version = "1.66"
links = "esp-idf-hal"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "hal", "esp-idf-sys/native"]
std = ["alloc", "esp-idf-sys/std", "esp-idf-sys/binstart", "embedded-svc?/std", "esp-idf-hal?/std", "esp-idf-svc?/std"]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = ">=0.34.1", default-features = false }
esp-idf-hal = { version = ">=0.43.1", optional = true, default-features = false }
esp-idf-svc = { version = ">=0.47.3", optional = true, default-features = false }
embedded-svc = { version = ">=0.26.4", optional = true, default-features = false }
embedded-hal = "0.2.7"

[build-dependencies]
embuild = "0.31.2"
```

I've tried to fix it with reading issue  #137, but without success.

Thanks for your help.

ivmarkov, 2024-12-14:
@emeric-martineau You are still on ESP-IDF V4.4.X and that's the root cause.
- A quick fix is to remove `espidf_time64` and instead add `espidf_time32`. See [this](https://github.com/rust-lang/libc/pull/3993) for more info.
- A longer-term fix is to upgrade your project to ESP-IDF 5.X (5.2.X is the latest version supported by the `esp-idf-*` Rust crates), as ESP-IDF 4.4.X is no longer supported.

emeric-martineau, 2024-12-14:
Thanks @ivmarkov ,

I try this to upgrade to ESP-IDF v5, but with v5.3.2 (and it's wrong).

closed 2024-12-14 by emeric-martineau