esp-rs/esp-idf-sys #359: Undefined reference to tinyusb callback funcs

fangpenlin, 2025-01-02:
## Bug description

Build extra `esp_tinyusb` component with `CONFIG_TINYUSB_MSC_ENABLED` set to `y`, somehow encounter the cakkback functions symbols missing issue:

```
error: linking with `ldproxy` failed: exit code: 1
  |
  = note: "ldproxy" "C:\\Users\\Fang-Pen\\AppData\\Local\\Temp\\rustcrQegWk\\symbols.o" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\esp32_playground-7d9086eaad0faf9f.2qoaz9wb7ep67z0qujfglu4yk.rcgu.o" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\esp32_playground-7d9086eaad0faf9f.4hds22qrruv73nrr9ljitkaw4.rcgu.o" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\esp32_playground-7d9086eaad0faf9f.ewjildy4dmea1mjhk07gswjqy.rcgu.o" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\esp32_playground-7d9086eaad0faf9f.ek3ed1swuuj50eabg05hzfn0u.rcgu.o" "-Wl,--as-needed" "-Wl,-Bstatic" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libesp_idf_svc-d89e248162f947f6.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembassy_futures-4dd7b98316da210e.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembassy_time_driver-8dad0d94c7b130f4.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libuncased-547947eb7e70b8f5.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libesp_idf_hal-02bb17ab7df2cca7.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_hal_async-e249d9e88b9180c5.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_can-919c04c9dd4a291c.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_hal-0da04245d18b2996.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libvoid-51cdb84cc35af7b3.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libnb-ed5b4bdf15f68b0e.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_hal_nb-ed5400dd423b5a44.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libnb-368281c0e611fbde.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libatomic_waker-8913311add159285.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembassy_sync-530069ec2ceaec36.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libfutures_util-d85c4e7fd26340f9.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libpin_project_lite-f0f5e50b49278681.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libfutures_task-ab077b771ac35733.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libpin_utils-b968344e217e695b.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libfutures_core-ea4b9ef6f4dcfee3.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libfutures_sink-75605a48403481da.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libcritical_section-4b666c255fceed72.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_hal-3d98d483e5d2907d.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libnum_enum-a233899e4d4d5925.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_svc-4783f60b49e30172.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libheapless-49b35f173bb6f03d.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libhash32-ab66924fc8ff433b.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libbyteorder-eefff2ac3abd8c9d.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libenumset-bdb60452e560eeb7.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_io_async-6e5224c4d6bc03db.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libembedded_io-ef685ca30757134c.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\liblog-cf2e205841b09931.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libesp_idf_sys-ab956bf0aa48d8ad.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\liblibc-105a914b80963923.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libconst_format-bfbe81f5d480c771.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libstd-4c468461821b8e42.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libpanic_abort-40b5e4816c6d2d03.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libobject-e2337356ce84003f.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libmemchr-7e638e6cd9a8bf3e.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libaddr2line-09108638b704a856.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libgimli-91179c1d341704e4.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\librustc_demangle-0df14b1482c9f94c.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libstd_detect-16421a88257927e7.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libhashbrown-f9fd6930d3df5877.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\librustc_std_workspace_alloc-225fd769b3c83fc9.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libminiz_oxide-68ad5c51c4d1a6a5.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libadler-1575b6ba46a624b7.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libunwind-b1e93e67cfecf252.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libcfg_if-bd93b87847037323.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\liblibc-696191a4e9a3bc43.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\liballoc-865a6e17b8f2eff0.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\librustc_std_workspace_core-851f43160a9ce72a.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libcore-372d31482e278cd7.rlib" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\libcompiler_builtins-52671fef70583e56.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-o" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\deps\\esp32_playground-7d9086eaad0faf9f" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "--ldproxy-linker" "C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe" "--ldproxy-cwd" "C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\build\\esp-idf-sys-d285e4edb12cb7fa\\out\\build" "@C:\\ws\\e3\\target\\xtensa-esp32s3-espidf\\debug\\build\\esp-idf-sys-d285e4edb12cb7fa\\out\\linker_args.txt"
  = note: [ldproxy] Running ldproxy
          Error: Linker C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe failed: exit code: 1
          STDERR OUTPUT:
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.proc_read10_cmd$constprop$0+0x8): undefined reference to `tud_msc_read10_cb'␍
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.mscd_xfer_cb+0x2c): undefined reference to `tud_msc_scsi_cb'␍
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.mscd_xfer_cb+0x30): undefined reference to `tud_msc_test_unit_ready_cb'␍
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.mscd_xfer_cb+0x34): undefined reference to `tud_msc_capacity_cb'␍
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.mscd_xfer_cb+0x38): undefined reference to `tud_msc_inquiry_cb'␍
          C:/ws/e3/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf\espressif__tinyusb\libespressif__tinyusb.a(msc_device.c.obj):(.literal.mscd_xfer_cb+0x3c): undefined refere [... cut ...]
The repository for reproduction is here:

https://github.com/LaunchPlatform/esp-idf-sys-esp-tinyusb-build-erro-reproduce

**Please note that to make the reproduction repo works, you need to downgrade espup to 11 with older clang runtime and also put the folder in a shorter path to avoid path too long problem under windows**

## Expected behavior

It should build

## Environment

- Crate (`esp-idf-sys`) version: 0.35.0
- ESP-IDF branch or tag: release/v5.2
- Target device (MCU): esp32s3
- OS: Windows11

fangpenlin, 2025-01-02:
I tried to understand why the build fail. It seems like that the esp-idf-sys build system is building tinyusb as a static lib?[ But the callback functions are defined in esp_tinyusb. May need to dig deeper to understand what's going on here.

ivmarkov, 2025-01-02:
> I tried to understand why the build fail. It seems like that the esp-idf-sys build system is building tinyusb as a static lib? But the callback functions are defined in esp_tinyusb. May need to dig deeper to understand what's going on here.

The callbacks are not really referenced from your program so the linker is pruning them. What happens if you [call this function](https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c#L172C21-L172C46) in your program?

fangpenlin, 2025-01-02:
@ivmarkov I think you're right. Indeed it seems like it's linker is pruning them.

I just call a random one from them `tud_msc_test_unit_ready_cb`:

```rust
unsafe { tud_msc_test_unit_ready_cb(0) };
```

And now it builds. Any idea how to make esp/rust build system to keep those functions without making a call? I can make a always false if condition to fix this problem as a workaround, but there must be a better solution than this.

ivmarkov, 2025-01-02:
> @ivmarkov I think you're right. Indeed it seems like it's linker is pruning them.
>
> I just call a random one from them `tud_msc_test_unit_ready_cb`:
>
> ```rust
> unsafe { tud_msc_test_unit_ready_cb(0) }; [...]

My point is that if you are trying to implement a mass-storage device, you have to place a call to [this function](https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/examples/peripherals/usb/device/tusb_msc/main/tusb_msc_main.c#L172C21-L172C46) _anyway_ in your code (right? or else how would your mass-storage device connect to the SD card?), and then if you do that, this function is anyway calling the callbacks, so they would be preserved?

fangpenlin, 2025-01-02:
@ivmarkov yep, I see, you're right. Looks like I will need to call that function anyway. So the header for `tusb_msc_storage` was missing in my code:

```
#include "tusb_msc_storage.h"
```

After that, if I call `tinyusb_msc_storage_mount` then the linker stop from removing the section.

Thank you so much for the help. Silly problem as it turns out 😅

closed 2025-01-02 by fangpenlin

ivmarkov, 2025-01-02:
BTW, I've now directly included all TinyUSB headers in the about-to-be-released next `esp-idf-sys` [here](https://github.com/esp-rs/esp-idf-sys/blob/ecf45d1b7f8c9acc8c55dbdd9b105c7ac51d8525/src/include/esp-idf/bindings.h#L575).

What this means is that while you still have to refer to the TinyUSB custom component from your binary crate, you no longer need to define your own "bindings_header" header. The moment you include the tinyusb component, the bindings will be generated for you.

Would you like to try this out? I really don't have time to try if it works correctly myself. You need to patch the `esp-idf-*` crates to their GIT versions in your `Cargo.toml` as follows:
```toml
[patch.crates-io]
embuild = { git = "https://github.com/esp-rs/embuild" }
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
```

Also note that you have to remove the `bindings_module` thing then, as these bindings are generated in the main `esp-idf-sys` namespace.

fangpenlin, 2025-01-02:
I just tried it out in my reproducing repo here:
https://github.com/LaunchPlatform/esp-idf-sys-esp-tinyusb-build-erro-reproduce/commit/0018f96f3f7d69d6c56c7eaacc192510e55dd66e

And here's the error I got

```
  Error: failed to generate bindings in file 'C:\ws\e3\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-35fce39328b4dadc\out\bindings.rs'
  Caused by:
      clang diagnosed error: C:/ws/e3/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-35fce39328b4dadc/out/managed_components/espressif__esp_tinyusb/include/tusb_cdc_acm.h:21:2: error: "TinyUSB CDC driver must be enabled in menuconfig"
  Stack backtrace:
```

Probably because in my reproducing repo, I only have `CONFIG_TINYUSB_MSC_ENABLED=y` instead of the flag for CDC

I guess you need to put `tusb_cdc_acm.h` header under `CONFIG_TINYUSB_CDC_ENABLED` flag

https://github.com/espressif/esp-usb/blob/e013116f2f6f1a71cda676f6438fed6d9d145b51/device/esp_tinyusb/CMakeLists.txt#L11C4-L11C30

probably the same for all the conditional stuff and their corresponding headers such as:

- CONFIG_TINYUSB_CDC_ENABLED
- CONFIG_TINYUSB_MSC_ENABLED
- CONFIG_TINYUSB_NET_MODE_NCM

ivmarkov, 2025-01-02:
> I just tried it out in my reproducing repo here: [LaunchPlatform/esp-idf-sys-esp-tinyusb-build-erro-reproduce@0018f96](https://github.com/LaunchPlatform/esp-idf-sys-esp-tinyusb-build-erro-reproduce/commit/0018f96f3f7d69d6c56c7eaacc192510e55dd66e)
> 
> And here's the error I got
> 
> ```
>   Error: failed to generate bindings in file 'C:\ws\e3\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-35fce39328b4dadc\out\bindings.rs'
>   Caused by:
>       clang diagnosed error: C:/ws/e3/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-35fce39328b4dadc/out/managed_components/espressif__esp_tinyusb/include/tusb_cdc_acm.h:21:2: error: "TinyUSB CDC driver must be enabled in menuconfig"
>   Stack backtrace:
> ```
> 
> Probably because in my reproducing repo, I only have `CONFIG_TINYUSB_MSC_ENABLED=y` instead of the flag for CDC
> 
> I guess you need to put `tusb_cdc_acm.h` header under `CONFIG_TINYUSB_CDC_ENABLED` flag
> 
> https://github.com/espressif/esp-usb/blob/e013116f2f6f1a71cda676f6438fed6d9d145b51/device/esp_tinyusb/CMakeLists.txt#L11C4-L11C30
> 
> probably the same for all the conditional stuff and their corresponding headers such as:
> 
> * CONFIG_TINYUSB_CDC_ENABLED
> * CONFIG_TINYUSB_MSC_ENABLED
> * CONFIG_TINYUSB_NET_MODE_NCM

Thank you for your feedback! Can you try again (do `cargo update` before re-building)
I think we only need to check for `CONFIG_TINYUSB_CDC_ENABLED`:
- The MSC header does not have any `CONFIG_*` checks
- The NET header does have a `CONFIG_*` check, [but it is of a different style](https://github.com/espressif/esp-usb/blob/20389182dbfd49f600f9fe56b70485f244b48087/device/esp_tinyusb/include/tinyusb_net.h#L14) (no compile-time error) and better (IMO) than what they did for the CDC header (?). Go figure! :-)

fangpenlin, 2025-01-02:
Hmmm 

That's odd. I tried your new commit

https://github.com/LaunchPlatform/esp-idf-sys-esp-tinyusb-build-erro-reproduce/commit/edc25d79944866674625810c48704e23514e0f71

Still encountering the same error

```
error: failed to run custom build command for `esp-idf-sys v0.35.0 (https://github.com/esp-rs/esp-idf-sys?rev=0bbcaab65ca1c199d21182a659184ae055cdd03d#0bbcaab6)`
Caused by:
  process didn't exit successfully: `C:\ws\e3\target\debug\build\esp-idf-sys-0feedbffd95c0f6e\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.11.2
  pip 24.3.1 from C:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\pip (python 3.11)
  Requirement already satisfied: pip in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (24.3.1)
  Requirement already satisfied: setuptools in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
  Using cached setuptools-75.6.0-py3-none-any.whl (1.2 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
      Found existing installation: setuptools 71.0.0
      Uninstalling setuptools-71.0.0:
        Successfully uninstalled setuptools-71.0.0
  Successfully installed setuptools-75.6.0
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Collecting setuptools (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 3))
    Using cached https://dl.espressif.com/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
  Requirement already satisfied: packaging in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 4)) (24.2)
  Requirement already satisfied: click in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 7)) (8.1.8)
  Requirement already satisfied: pyserial in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 8)) (3.5)
  Requirement already satisfied: cryptography in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 9)) (41.0.7)
  Requirement already satisfied: pyparsing in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 10)) (3.2.1)
  Requirement already satisfied: pyelftools in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 11)) (0.31)
  Requirement already satisfied: idf-component-manager in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.1.1)
  Requirement already satisfied: esp-coredump in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 13)) (1.12.0)
  Requirement already satisfied: esptool in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 14)) (4.8.1)
  Requirement already satisfied: esp-idf-kconfig in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 15)) (1.5.0)
  Requirement already satisfied: esp-idf-monitor in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 16)) (1.5.0)
  Requirement already satisfied: esp-idf-size in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 17)) (1.6.1)
  Requirement already satisfied: esp-idf-panic-decoder in c:\ws\e3\.embuild\espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from -r C:\ws\e3\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 18)) (1.2.1)
[... cut ...]
    10: std::rt::lang_start<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at C:\actions-runner\_work\rust-build\rust-build\rust\library\std\src\rt.rs:163
    11: main
    12: invoke_main
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    13: __scrt_common_main_seh
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    14: BaseThreadInitThunk
    15: RtlUserThreadStart
Process finished with exit code 101

```

ivmarkov, 2025-01-02:
OK, let me try myself, though weird - as if you did not pick up the latest commit.

ivmarkov, 2025-01-02:
Don't do `rev = `, you are this way actually not applying the `esp-idf-sys` crate as a patch.

fangpenlin, 2025-01-02:
Yep, I got it wrong. Didn't notice that the patch actually was not applied. Sorry still pretty new to rust 😅

And yes, I tested it locally now it builds. 👍

Thanks for the help! 🙌

ivmarkov, 2025-01-02:
> Yep, I got it wrong. Didn't notice that the patch actually was not applied. Sorry still pretty new to rust 😅
> 
> And yes, I tested it locally now it builds. 👍
> 
> Thanks for the help! 🙌

Thank you for the help and no you did not get it wrong (that's why I did strikethrough my original message).

It wasn't working because the tiny usb config settings (for whatever reason) are always present, but are either equal to 0 or to 1, so I had to fix the `#idfef` to just `#if`.