esp-rs/esp-idf-sys #363: Add option to skip tool downloading entirely

Janrupf, 2025-01-19:
## Motivations

I'm running NixOS and installing the tools via the espidf own methods fails. Since I'm already providing the matching tools externally, the build succeeds as soon as I patch out the tool installation.

## Solution

Add an environment, something like `ESP_IDF_SYS_SKIP_TOOL_INSTALL` to skip installing the tools and rely on the user to provide them.

## Alternatives

1. Just don't support using external tools - this more or less breaks the build on NixOS and requires ugly workarounds (FHS environments for example)
2. Very explicitly support NixOS and run the tooling NixOS provides to patch the binaries to run successfully

I believe both of these have major drawbacks, while providing an environment variable is a simple and concise solution.

## Additional context

NixOS can't run Linux binaries downloaded straight from the internet due to the way dynamic library linkage works on NixOS. There are workarounds, but in this case providing the tools externally is reasonably.

If wanted, I can provide the necessary PR to implement these changes.

Vollbrecht, 2025-01-19:
You can always provide your own env, its documented in the docs. You need a fully activated esp-idf env. What that means is described in the offical esp-idf guides. For not using the automated install look at this [option](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir)'s

Janrupf, 2025-01-19:
I'm aware that I can in theory use pre-activated esp-idf environment, however, when just disabling the tool downloads via an environment variable I can still take advantage of the automatic idf download. This should not be a problem provided the external tools match, and if not the idf complains loudly (and I'd assume a user forcing external tools knows how to deal with that error message).

ivmarkov, 2025-01-19:
> I'm aware that I can in theory use pre-activated esp-idf environment, however, when just disabling the tool downloads via an environment variable I can still take advantage of the automatic idf [...]

1. The "automatic idf download" that you would like to take advantage of is equivalent to this one-liner: 
```sh
git clone --recurse-submodules https://github.com/espressif/esp-idf
```

... so I'm not sure you would be automating that much

2. Pre-installing the tools for some ESP IDF version _before_ cloning the corresponding ESP IDF version tag/branch from GIT is backwards to what you should usually do, as the tooling is very specific to the concrete ESP IDF version which is being cloned. You are supposed to (a) first clone/ GIT checkout the ESP IDF branch/tag, and then (b) install the tooling by using the `install.sh` script which is _inside that particular ESP IDF version_. Otherwise you would be doing guesswork as to what tooling you actually need.

===

Not using nixos myself, but a lot of folks actually did complain in the past that the `esp-idf-sys` build assumes that the ESP IDF is a GIT repo (apparently, they have some ways to download the ESP IDF in some sort of nixos package or whatever which is not a GIT repo). To support this use-case, I even merged PRs against `embuild` and `esp-idf-sys` recently. Now, you seem to want exactly the opposite...

===

Also in general, I'm not so happy with putting additional options knobs and whatnot to what is already a very, very complex build. I think we should stick to the existing three options:
- You provide both tooling and the ESP IDF, activate the environment these and then `esp-idf-sys` just uses those. Also helps with security sensitive folks who might want to do vendoring of the ESP IDF
- You provide an ESP IDF directory or a custom GIT url, and then `esp-idf-sys` uses it, but downloads the necessary tooling (BTW: if you use the user-local rather than the crate-local directory for the tooling and pre-download the tooling there, `esp-idf-sys` will download nothing or only the missing parts)
- You tell `esp-idf-sys` what ESP IDF version, and it does the necessary

ivmarkov, 2025-01-23:
No reply, closing

closed 2025-01-23 by ivmarkov

Janrupf, 2025-01-24:
Oh sorry, yeah, forgot about this. The explanation you have provided is fair, I'll switch to providing the ESP-IDF externally (since my argument of having everything version locked makes more sense then anyway) - I wasn't aware of the really tight coupling between tool versions and esp-idf versions.