esp-rs/esp-idf-sys #364: Apply and persist required esp-idf changes for esp_config.h files

chris-subtlebytes, 2025-01-19:
## Motivations

Some files in the ESP IDF *require* local modifications because they don't have sdkconfig options for them.

In my case, I need to enable `MBEDTLS_USE_PSA_CRYPTO`, which can only be enabled by patching:
`components/mbedtls/port/include/mbedtls/esp_config.h`:

```
#define MBEDTLS_USE_PSA_CRYPTO
```

There's also some broken sdkconfig option combinations that need some local patching for now and eventually upstream fixes. In my case, some TLS1.2 macros still try to compile when the TLS1.2 config is disabled.

As of now, there's no official way to automate patching or persist these changes reliably. If these patches are committed, then the cargo build system will actually erase the changes with a fresh sync of the esp idf remote. The current workaround is to manually apply them and leave the `.embuild/espressif/esp-idf/v*` repo dirty.

- Would you like to implement this feature? [y/n]

No

## Desired behavior

A user can download an `esp-idf-sys` based repository that requires patches to `esp-idf` and upon `cargo build`, it automatically syncs the `esp-idf` (as it does today, or by explicit syncs as some other open FRs suggest), then applies the required patches on top so that the build is successful.

## Alternatives

## Additional context

ivmarkov, 2025-01-23:
Sorry for replying with a delay.

I'm are actually trying to escape the notion of "patches", and - since quite some ESP-IDF releases - we don't have to apply patches anymore. The last release which needs patches is V5.0.X (and the no longer officially supported 4.4.X), and that's only if you would like to use the `esp_app_desc!` macro.

Back in the days, the `ESP-IDF` build was based on PlatformIO (feature `pio` which is still around, but no longer the default, and I bet not used by anyone by now).

With PIO, we did not have any meaningful alternative, as the PIO build does not clone an ESP-IDF GIT repo, but rather, downloads the ESP-IDF as ZIP or so.

These days what I would rather suggest is the most natural thing, which is - if you need your custom changes, then just fork the ESP-IDF repo, make your changes in your own fork, and then instruct `esp-idf-sys` to use your fork rather than the official ESP-IDF GIT repo & branches.

And the above is a necessary setup for PR-ing/upstreaming the changes, so you would need it anyway if you plan to upstream those.

chris-subtlebytes, 2025-02-19:
> The last release which needs patches is V5.0.X

I took another look around for a way to #define MBEDTLS_USE_PSA_CRYPTO but the only way I see is by modifying the tree itself at `components/mbedtls/port/include/mbedtls/esp_config.h` even on the esp-idf master branch.

Because it's just a define, is there a way to pass a cflag arg to esp-idf-sys? I poked around cargo_driver.rs but didn't see anything useful.

I'd rather not maintain a fork just to pass a flag. The TLS 1.2 compilation issues I can work around just by enabling it while upstreaming some fixes.

Thank you for your time.

ivmarkov, 2025-02-20:
[By looking at one of the ESP-IDF native examples](https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/protocols/https_mbedtls/main/https_mbedtls_example_main.c#L65) it seems to me that putting `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y` in your `sdkconfig.defaults` should already enable PSA CRYPTO (as TLS 1.3 anyway depends on it, as far as I remember)?

chris-subtlebytes, 2025-02-22:
Setting that config doesn't `#define MBEDTLS_USE_PSA_CRYPTO` at any point in the build process though so some of the external-facing functions aren't exposed. I'll just make the fork for now and see if I can get upstream to accept plumbing it through sdkconfig.