esp-rs/esp-idf-sys #366: Unable to build on exFAT on Linux

untbu, 2025-01-23:
While building, esp-idf automatically creates a python venv. The problem is that on Linux, this venv uses symlinks (python has an option to turn that off, but it doesn't work...), but exFAT does not support symlinks.
It would therefore be good to mention this requirement for symlinks in the README (or maybe the README of esp-idf-template?).

ivmarkov, 2025-01-23:
I do not understand your issue. What does the python venv has to do with exFAT supporting or not supporting symlinks?

Are you trying to place the build artefact and your project on exFAT?

untbu, 2025-01-23:
My project and the `.embuild`-folder and therefore also the venv in `.embuild/espressif/python_env/idf5.2_py3.10_env` are on exFAT and when building the example I always get this error:

```
Error: [Errno 1] Operation not permitted: 'lib' -> '/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/python_env/idf5.2_py3.10_env/lib64'
  Traceback (most recent call last):
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2841, in <module>
      main(sys.argv[1:])
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2833, in main
      action_func(args)
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2239, in action_install_python_env
      subprocess.check_call([sys.executable, '-m', 'venv',
    File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'venv', '--clear', '/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/python_env/idf5.2_py3.10_env']' returned non-zero exit status 1.
  Error: Could not install esp-idf
```

<details>
<summary>complete log</summary>

```
   Compiling compiler_builtins v0.1.143
   Compiling object v0.36.7
   Compiling proc-macro2 v1.0.93
   Compiling rustc-std-workspace-core v1.99.0 (/home/untbu/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling libc v0.2.169
   Compiling serde v1.0.217
   Compiling aho-corasick v1.1.3
   Compiling crossbeam-utils v0.8.21
   Compiling clang-sys v1.8.1
   Compiling rustix v0.38.44
   Compiling rustversion v1.0.19
   Compiling bstr v1.11.3
   Compiling prettyplease v0.2.29
   Compiling syn v1.0.109
   Compiling serde_json v1.0.137
   Compiling either v1.13.0
   Compiling nix v0.29.0
   Compiling linux-raw-sys v0.4.15
   Compiling thiserror v1.0.69
   Compiling walkdir v2.5.0
   Compiling libloading v0.8.6
   Compiling nom v7.1.3
   Compiling cvt v0.1.2
   Compiling once_cell v1.20.2
   Compiling ryu v1.0.18
   Compiling camino v1.1.9
   Compiling bindgen v0.71.1
   Compiling autocfg v1.4.0
   Compiling heck v0.4.1
   Compiling std v0.0.0 (/home/untbu/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
   Compiling semver v1.0.25
   Compiling itoa v1.0.14
   Compiling anyhow v1.0.95
   Compiling itertools v0.13.0
   Compiling cc v1.2.10
   Compiling normpath v1.3.0
   Compiling rustc-hash v2.1.0
   Compiling home v0.5.11
   Compiling bitflags v1.3.2
   Compiling fastrand v2.3.0
   Compiling iana-time-zone v0.1.61
   Compiling quote v1.0.38
   Compiling unicode-xid v0.2.6
   Compiling alloc v0.0.0 (/home/untbu/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/alloc)
   Compiling cfg-if v1.0.0
   Compiling crossbeam-epoch v0.9.18
   Compiling unwind v0.0.0 (/home/untbu/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/unwind)
   Compiling adler2 v2.0.0
   Compiling memchr v2.7.4
[... 225 lines cut ...]
  Error: [Errno 1] Operation not permitted: 'lib' -> '/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/python_env/idf5.2_py3.10_env/lib64'
  Traceback (most recent call last):
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2841, in <module>
      main(sys.argv[1:])
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2833, in main
      action_func(args)
    File "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py", line 2239, in action_install_python_env
      subprocess.check_call([sys.executable, '-m', 'venv',
    File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'venv', '--clear', '/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/python_env/idf5.2_py3.10_env']' returned non-zero exit status 1.
  Error: Could not install esp-idf

  Caused by:
      command 'env -u IDF_PYTHON_ENV_PATH -u MSYSTEM IDF_TOOLS_PATH="/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif" "python3" "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3/tools/idf_tools.py" "--idf-path" "/media/untbu/Ventoy/Projects/esp-idf-sys/.embuild/espressif/esp-idf/v5.2.3" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

</details>

Building the exact same code on ext4 works. I also tried to create a venv on exFAT manually, but it doesn't work. So I guess it's a general python problem. I just didn't expect a rust project to not work on exFAT and I don't want others to waste their time.

Build command:
```
CARGO_TARGET_DIR=target MCU=esp32c3 cargo +nightly build --target riscv32imc-esp-espidf --example std_basics
```

ivmarkov, 2025-01-23:
I think the ESP IDF build itself likely does not support building on "exFAT". If you would like this mentioned in `esp-idf-sys`, open a small PR about that. But we'll likely never fix that. One workaround you can try is to set the ESP_IDF_INSTALL_DIR to "global", i.e. to your home directory. This way the symlinks will not be created on the exFAT but in your native linux partition.

PR #367 (2025-01-23): Add note about FAT incompatibility to README
ESP IDF can’t be compiled on filesystems without support for symbolic links because it creates a virtual python venv during build which requires symlinks (at least on Linux).

Closes #366