esp-rs/esp-idf-sys #371: Unable to build with extra component esp-camera

Geostartico, 2025-02-12:
## Bug description

I have been trying to build the template generated by esp-generate by only adding the necessary code in the toml to compile with the extra component esp-camera, however it doesn't find the xtensa toolchain

## To Reproduce

cargo.toml:
```
[package]
edition = "2021"
name    = "rust_camera_extra_component"
version = "0.1.0"

[[bin]]
name = "rust_camera_extra_component"
path = "./src/bin/main.rs"

[dependencies]
critical-section = "1.2.0"
esp-backtrace = { version = "0.15.0", features = [
  "esp32",
  "exception-handler",
  "panic-handler",
  "println",
] }
esp-hal = { version = "0.23.1", features = ["esp32", "unstable"] }
esp-idf-sys = { version = "0.36.1", default-features = false, features = ["binstart", "native"]}
esp-println = { version = "0.13.0", features = ["esp32", "log"] }
log = { version = "0.4.21" }

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units    = 1     # LLVM can perform better optimizations using a single thread
debug            = 2
debug-assertions = false
incremental      = false
lto              = 'fat'
opt-level        = 's'
overflow-checks  = false

[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "espressif/esp32-camera", version = "2.0.7" }
bindings_header = "./bindings.h"
bindings_module = "camera"

```
.cargo/config.toml:
```
[target.xtensa-esp32-espidf]
runner = "espflash flash --monitor"
linker = "ldproxy"

[env]
ESP_LOG="INFO"

[build]
rustflags = [
  "-C", "link-arg=-nostartfiles",
  "--cfg", "espidf_time64",
  "-C", "default-linker-libraries",
]

target = "xtensa-esp32-espidf"

[unstable]
build-std = ["core"]
[... 200 lines cut ...]
    23:     0x7617cff5d488 - <unknown>
    24:     0x7617cff5d54c - __libc_start_main
    25:     0x5b91f3170f35 - _start
    26:                0x0 - <unknown>
warning: build failed, waiting for other jobs to finish...

```

## Environment
ESP-IDF v5.3
toolchain installed from through espup
- Target device (MCU): esp32 (in the path the following commands for xtensa-esp32 are present:
 xtensa-esp32-elf-addr2line   xtensa-esp32-elf-gcc-14.2.0  xtensa-esp32-elf-lto-dump  
xtensa-esp32-elf-ar          xtensa-esp32-elf-gcc-ar      xtensa-esp32-elf-nm        
xtensa-esp32-elf-as          xtensa-esp32-elf-gcc-nm      xtensa-esp32-elf-objcopy   
xtensa-esp32-elf-c++         xtensa-esp32-elf-gcc-ranlib  xtensa-esp32-elf-objdump   
xtensa-esp32-elf-cc          xtensa-esp32-elf-gcov        xtensa-esp32-elf-ranlib    
xtensa-esp32-elf-c++filt     xtensa-esp32-elf-gcov-dump   xtensa-esp32-elf-readelf   
xtensa-esp32-elf-cpp         xtensa-esp32-elf-gcov-tool   xtensa-esp32-elf-size      
xtensa-esp32-elf-elfedit     xtensa-esp32-elf-gdb         xtensa-esp32-elf-strings   
xtensa-esp32-elf-g++         xtensa-esp32-elf-gprof       xtensa-esp32-elf-strip     
xtensa-esp32-elf-gcc         xtensa-esp32-elf-ld                                     
xtensa-esp32-elf-gcc-13.2.0  xtensa-esp32-elf-ld.bfd     )                            

- OS: EndeavourOS x86_64 Linux 6.13.1-zen3-1-zen

Vollbrecht, 2025-02-12:
Don't mix esp-hal and esp-idf-sys. They are separate projects, e.g esp-hal is standalone baremetall, and esp-idf-sys is running on-top of FreeRTOS using the esp-idf sdk.

Geostartico, 2025-02-12:
> Don't mix esp-hal and esp-idf-sys. They are separate projects, e.g esp-hal is standalone baremetall, and esp-idf-sys is running on-top of FreeRTOS using the esp-idf sdk.

ok but even removing the esp-hal dependency doesn't solve the issue

Vollbrecht, 2025-02-12:
same goes for esp-println and esp-backtrace. Also what is your reasoning for providing nostartfiles and default linker libraries as rustflags?

Vollbrecht, 2025-02-12:
Ah you did not use the esp-idf-template in the first place but the esp generate tool? Yeah please use the esp-idf-template as a base. Than just adding the esp-camera module should get you the bindings.

Geostartico, 2025-02-13:
Yea that was the issue, thanks for the help. just out of pure curiosity, how do i know which function is std-only in esp-idf-svc?

closed 2025-02-13 by Geostartico

ivmarkov, 2025-02-13:
> just out of pure curiosity, how do i know which function is std-only in esp-idf-svc?

It does not make any sense to use `esp-idf-svc` without STD. If you use `esp-idf-svc`, then you can assume STD is available, and then _that's actually the way_ to use threads, network IO and filesystem IO in your Rust firmware (via the Rust STD API).

`alloc` is also available (Strings, Vecs and such) but you need to be careful not to overuse it as that will bring unpredictability in your firmware.

What is **NOT** allowed / wrong is to mix the `esp-idf-svc` HAL (and its companion crates, `esp-idf-hal` and `esp-idf-sys`) - which sit on top of the ESP IDF C framework - with the "baremetal" `esp-hal` HAL, and with its companion crates, like `esp-println`, `esp-backtrace` and `esp-wifi` which do NOT sit on top of the ESP-IDF C framework and operate the hardware directly.

You have to choose ONE framework (ESP-IDF, and thus STD) OR the other (no ESP-IDF and no STD as `esp-hal` has no STD support, so multitasking, network IO and filesystem IO are done using other means), but you cannot use both (for now, and likely forever).

Usually choosing one or the other is deciding on the trade-off between having more drivers or a need to interop with an existing C codebase (ESP-IDF) vs more "pure Rust" and less memory and flash size consumption (`esp-hal`). `esp-hal` is also an Espressif-sponsored project with a bunch of FTEs assigned to it, while `esp-idf-*` is a community project.

Once you choose ESP-IDF or baremetal however, you can still use most of the generic crates from the Embassy echosystem if you want to go async. As well as (for ESP-IDF specifically) some low-footprint generic STD-only crates.

Using generic no_std crates like - say - `serde_json_core` is possible with both of course.