esp-rs/esp-idf-sys #375: Cannot compile in macOS

Adancurusul, 2025-03-28:
The same error as  [#287,](https://github.com/esp-rs/esp-idf-sys/issues/287) but neither of the methods can be solved.

My device is a m2 macbook air

Is it because I have done something wrong?

```
[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }
esp-idf-hal = { version = "0.43", default-features = false }
esp-idf-sys = { version = "0.34", default-features = false }
esp32-nimble = "0.6.0"
```

When I use

```
[patch.crates-io]
cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
```

When the compilation still reports the same error and there is warning

```
warning: Patch `cc v1.0.83 (https://github.com/rust-lang/cc-rs?tag=1.0.83#bfc3ba41)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
```

If using

```
export CRATE_CC_NO_DEFAULTS=1
```

esp-idf-svc will report an error

```
   Compiling esp32-nimble v0.6.1
   Compiling ble-serial-bridge v0.1.0 (/Users/hondachen/codes/esp32/rs_code/ble-serial-bridge)
error[E0308]: mismatched types
   --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/tls.rs:212:36
    |
212 |                 rcfg.alpn_protos = bufs.alpn_protos.as_mut_ptr();
    |                 ----------------   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*mut *const u8`, found `*mut *const i8`
    |                 |
    |                 expected due to the type of this binding
    |
    = note: expected raw pointer `*mut *const u8`
               found raw pointer `*mut *const i8`

error[E0308]: mismatched types
     --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/tls.rs:441:25
      |
440   |                     sys::esp_tls_conn_new_async(
      |                     --------------------------- arguments to this function are incorrect
441   |                         host.as_bytes().as_ptr() as *const i8,
      |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*const u8`, found `*const i8`
      |
      = note: expected raw pointer `*const u8`
                 found raw pointer `*const i8`
note: function defined here
     --> /Users/hondachen/codes/esp32/rs_code/ble-serial-bridge/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-25b9b84a547c99dd/out/bindings.rs:41051:12
      |
41051 |     pub fn esp_tls_conn_new_async(
      |            ^^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
     --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/tls.rs:449:25
      |
448   |                     sys::esp_tls_conn_new_sync(
      |                     -------------------------- arguments to this function are incorrect
449   |                         host.as_bytes().as_ptr() as *const i8,
      |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*const u8`, found `*const i8`
      |
      = note: expected raw pointer `*const u8`
                 found raw pointer `*const i8`
note: function defined here
     --> /Users/hondachen/codes/esp32/rs_code/ble-serial-bridge/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-25b9b84a547c99dd/out/bindings.rs:41033:12
      |
41033 |     pub fn esp_tls_conn_new_sync(
      |            ^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
   --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/private/cstr.rs:142:8
    |
137 |         cstrs[i] = cstr.as_ptr();
    |         ----- here the type of `cstrs` is inferred to be `[*const u8; N]`
...
142 |     Ok(cstrs)
    |     -- ^^^^^ expected `[*const i8; N]`, found `[*const u8; N]`
    |     |
    |     arguments to this enum variant are incorrect
    |
    = note: expected array `[*const i8; N]`
               found array `[*const u8; N]`
help: the type constructed contains `[*const u8; N]` due to the type of the argument passed
   --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/private/cstr.rs:142:5
    |
142 |     Ok(cstrs)
    |     ^^^-----^
    |        |
    |        this argument influences the type of `Ok`
note: tuple variant defined here
   --> /Users/hondachen/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/result.rs:532:5
    |
532 |     Ok(#[stable(feature = "rust1", since = "1.0.0")] T),
    |     ^^

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-svc` (lib) due to 4 previous errors
```

ivmarkov, 2025-03-28:
Also:

> ```
> [dependencies]
> log = { version = "0.4", default-features = false }
> esp-idf-svc = { version = "0.48", default-features = false }
> esp-idf-hal = { version = "0.43", default-features = false }
> esp-idf-sys = { version = "0.34", default-features = false }
> esp32-nimble = "0.6.0"
> ```

Way too old! Upgrade to latest-released versions of these crates.

> ```
> [patch.crates-io]
> cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
> ```

Not necessary with the latest-released `esp-idf-*` crates. 
Why don't you generate a new binary crate with `esp-idf-template`, use the `esp`  toolchain, as explained above, and start from there?

> When the compilation still reports the same error and there is warning
>
> ```
> warning: Patch `cc v1.0.83 (https://github.com/rust-lang/cc-rs?tag=1.0.83#bfc3ba41)` was not used in the crate graph. [...]

Not necessary to patch the CC-RS crate. The necessary changes there were upstreamed in it several months ago.

> If using
> 
> ```
> export CRATE_CC_NO_DEFAULTS=1
> ```
> 
> esp-idf-svc will report an error
> 
> ```

Not necessary either.

> error[E0308]: mismatched types
>    --> /Users/hondachen/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.48.1/src/tls.rs:212:36
>     |
> 212 |                 rcfg.alpn_protos = [...]

Upgrade.

closed 2025-03-28 by ivmarkov

Estebanrg21, 2025-06-15:
So this issue still exists.
I'm not even able to compile the basic example of the [esp-idf-template](https://github.com/esp-rs/esp-idf-template) on a m2.
I'm obtaining the following error:
```
Caused by:
  process didn't exit successfully: `/path/to/target/debug/build/esp-idf-sys-6e6049ec985d1465/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'aarch64-apple-darwin'
warning: build failed, waiting for other jobs to finish...
```

This was present using the rust toolchain v1.85 and esp-idf-svc = v0.51.

I tried to do what I could understand from the response of @ivmarkov which consists in downgrading the rust toolchain to 1.85 and downgrade the only dependency of `esp-idf-*`, which is `esp-idf-svc`, from v0.51 to v0.48. This didn't work as it now shows another error that goes as follows:
```
   Compiling embassy-futures v0.1.1
   Compiling project-name v0.1.0 (/path/to/project)
error[E0433]: failed to resolve: could not find `espidf` in `embuild`
  --> build.rs:2:14
   |
2  |     embuild::espidf::sysenv::output();
   |              ^^^^^^ could not find `espidf` in `embuild`
   |
note: found an item that was configured out
  --> /path/to/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embuild-0.33.0/src/lib.rs:24:9
   |
24 | pub mod espidf;
   |         ^^^^^^
note: the item is gated behind the `espidf` feature
  --> /path/to/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embuild-0.33.0/src/lib.rs:23:7
   |
23 | #[cfg(feature = "espidf")]
   |       ^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0433`.
error: could not compile `project-name` (build script) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
```
Is there any other workaround?

ivmarkov, 2025-06-15:
@Estebanrg21 Your issue has nothing to do with the original problem here, aside from the fact that you also happen to try on MacOS.

The root cause is actually listed in the error log:
```
Error: Unsupported target 'aarch64-apple-darwin'
```

Two likely reasons for that:
1 - You've touched the `.cargo/config.toml` of the generated project and changed the target there to the target for your mac. Don't do that - you are not compiling _for_ your mac, but for the MCU. 
2 - You have a [`CARGO_BUILD_TARGET` env var or a `RUSTC` env var or similar](https://doc.rust-lang.org/cargo/reference/environment-variables.html). Using such env vars is a very, very bad idea, so remove everything Rust-related from your environment vars, as well as everything C-compiler related, like `CXX`, `CFLAGS` etc. etc.

ivmarkov, 2025-06-15:
> Hey [@ivmarkov](https://github.com/ivmarkov) thanks for the quick response!
> 
> So apparently, none of the reasons you listed were the root cause of the issue having that I didn't touch the `.cargo/config.toml` and didn't define that env var either. I had to redo the creation of the project, and instead of selecting v5.3 for the ESP-IDF version I selected v5.2 and it worked.

Sounds quite unbelievable to me but since you have so.ething working - OK fine